chipsalliance/caliptra-mcu-sw #399: Caliptra MCU RT - Time To Ready (TTR) impact on HPKE algorithms

mgbharath, 2025-09-04:
OCP LOCK requires HPKE Key pairs to be generated during the cold boot. Spec plans to support three variants of HPKE algorithms

DHKEM(P-384, HKDF-SHA384) (0x0011), AES-256-GCM (0x0002), HKDF-SHA384 (0x0002)
ML-KEM-1024 (0x0042), AES-256-GCM (0x0002), HKDF-SHA384 (0x0002)
QSF-KEM(ML-KEM-1024,P-384)-XOF(SHAKE256)-KDF(SHA3-256) (0x0a25), AES-256-GCM (0x0002), HKDF-SHA384 (0x0002)

It may also require performing Known Answer Test (KAT) for underlaying primitives used by HPKE : ECDH, AES, KDF

How much time it takes to complete self-tests and key generation all the three variants?

JohnTraverAmd, 2025-11-19:
For the primitives, we do have KATs that are executing. HPKE does not require additional KATs.

mgbharath, 2025-11-24:
@JohnTraverAmd  Thanks for the response.

closed 2025-11-24 by mgbharath