chipsalliance/caliptra-mcu-sw #402: Can Caliptra DMA be uses for SoC FW measurement/loading?

elecharo, 2025-09-04:
In Caliptra 2.0 main specification the flow to load SoC FW is described in "Common Run-Time Authentication Flows":

"1. MCU RT FW will do PLDM T5 flow to obtain the FW images for downstream uControllers (or other SoC configuration)
2. MCU RT FW will send/stream the (FW or config) payload to Caliptra SHA Acc to perform hash measurements as the payload comes through the MCTP transport.
3. MCU RT FW can either stage the entire image or write to the final destination as a part of the previous step depending on the SOC construction.
Note: By SOC security/design construction, the FW/payload that is loaded must NOT be allowed to execute or be used until Caliptra authorizes that the FW/payload.
4. MCU RT FW will issue the imageID & GO-field bit (bit that Caliptra RT FW would set if the image authorization is successful) to Caliptra RT FW to start off the process of image authorization of the image that was hashed."

Is there an alternate flow that makes use of Caliptra's DMA to measure and authenticate SoC FW available in a SRAM outside Caliptra? Caliptra's DMA is used to load MCU FW to MCI SRAM, can it be used for SoC FW (DMA reads from SoC SRAM, measures and authenticates FW without need for MCU to copy it into Caliptra's mailbox).

swenson, 2025-12-05:
I believe we do use DMA for this. @mlvisaya can you confirm?

mlvisaya, 2025-12-05:
Yes, this can be achieved using the AUTHORIZE_AND_STASH command with the `source` field set to `LoadAddress` (https://github.com/chipsalliance/caliptra-sw/blob/main/runtime/README.md#authorize_and_stash).

The AXI DMA Address (Image Load Address) of the SoC Image needs to be set first in the Auth Manifest (see https://github.com/chipsalliance/caliptra-sw/tree/main/auth-manifest#image-metadata-entry).

When the AUTHORIZE_AND_STASH command with `LoadAddress` set as source is handled, Caliptra will retrieve the LoadAddress in the Auth Manifest for the particular SoC Image. Caliptra will then DMA the SoC Image from the `LoadAddress` into SHA to authorize it and compare the digest to the one in the Auth Manifest. 

This actually being done right now by the Image Loading module of MCU SDK.

closed 2026-01-30 by mlvisaya